betavr/steamvr_driver_hydra #2: VR Server crashes with no Razer Hydra present

cbuchner1, 2016-04-19:
The driver makes the VR Server process crash when no Razer Hydra is connected to the machine.
There's no difference if I use your build, or a build I made myself.

I get this in the log message of VR Server, and after that it's crashing:
Wed Apr 20 2016 01:16:48.768 - Driver hydra has no HMDs. Skipping

Note that my machine never had a Razer Hydra connected and there is no driver installed for it either.

betavr, 2016-04-20:
That's the last line in the log when after it crashes? Are you using the latest SteamVR beta?

cbuchner1, 2016-04-20:
I wasn't on SteamVR beta. But even after switching to the current beta release, it keeps crashing. When I remove the driver, it runs again. I shall use a debug build to see if I can attach a debugger to SteamVR.

cbuchner1, 2016-04-20:
```
std::thread *m_Thread;   // change this to a pointer

CServerDriver_Hydra::CServerDriver_Hydra()
	: m_Thread(NULL),   // added this
	  m_bStopRequested( false )
	, m_bLaunchedHydraMonitor( false )
{
}

void CServerDriver_Hydra::Cleanup()
{
	if ( m_Thread && m_Thread->joinable() )  // test for NULL m_Thread
	{
		m_bStopRequested = true;
		m_Thread->join();
		m_Thread = NULL;  // force to NULL after thread join()
		sixenseExit();
	}
}
```

WIth these code changes the crashes are gone.

NOTE: Cleanup() is getting called multiple times when the VR Server is shutting down

NOTE: Running a debug version required the Visual Studio 2010 debug runtimes (installing Visual Studio 2010, meh) and the DeviceDLL.dll found in the sixense_simple3d example folder.

betavr, 2016-04-20:
The new operator before std::thread is giving me an error, how did you successfully compile it? Also the thread should do the cleaning up in the destructor, maybe that's not called on your system for some other reason. Have you tried other SteamVR builds, like one of the earlier betas? The latest one is giving me problems too (can't open the dashboard).

cbuchner1, 2016-04-21:
When the new operator fails, you probably didn't replace the m_Thread member in the CServerDriver_Hydra header file with a pointer

Now the root-cause of the crashes is as follows

The Visual C++ project called this when building the driver beause I had set the driver's installation folder to my SteamVR drivers folder!

vrpathreg adddriver "D:\Steam\steamapps\common\SteamVR\drivers\leap"

However steam is already looking for drivers in its drivers folder. So due to the adddriver command the leap driver got loaded twice by steam - somehow lead to a duplicate shutdown of the driver when no controllers were located.

My above outlined fix still prevents the crash from occuring in this situation. But so does a:

vrpathreg removedriver "D:\Steam\steamapps\common\SteamVR\drivers\leap"

closed 2016-04-21 by cbuchner1

betavr, 2016-04-21:
Thanks for the heads up!

Relys, 2016-04-22:
I had to apply the same thread pointer fixes to get my build to stop crashing.

betavr, 2016-04-22:
Have you tried removind the driver with vrpathreg first?