betavr/steamvr_driver_hydra #3: [Feature request] Mount Hydra base on head.

sigboe, 2016-04-20:
Would be nice if the driver supported the possibility to have the hydra mounted on the head. Its one step closer to room scale 😄 And the DK2 has a USB port so its got that going for it. You can even run the cords to the controllers inside your shirt. A little duct tape and its mounted to the DK2 strap or the earphones.

betavr, 2016-04-20:
Good idea. Need to look into how to access the hmd's pose to calculate the relative position.

cbuchner1, 2016-04-24:
The driver_leap repository that I forked off Valve's driver_hydra repo contains code to do the tracking relative to the HMD. This is because the Leap Motion is usually attached to the HMD.

betavr, 2016-04-24:
Will take a look at that, thanks!

CBerry22, 2016-05-26:
Seconding this, the ability to mount on the head would massively increase the hydra's usability in games. Specifically games with 360 movement.

sigboe, 2016-06-09:
@betavr Dod you have a look into @cbuchner1 potential solution? 

Sorry for being inpatient :stuck_out_tongue: It's just room scale is so awesome.

betavr, 2016-06-09:
I did, but haven't had time to finish it yet. Will give it higher priority, but first there will be an update regarding the install method (actually no installer will be needed).

arronhuk, 2016-06-10:
Would probably be best to allow it to ignore head orientation too so you can install the base on your body rather than on the head. I tried to play job simulator but the cables wouldn't stretch far enough to set up a room

cbuchner1, 2016-06-10:
And who's tracking your Body in the SteamVR coordinate space?

arronhuk, 2016-06-10:
I was thinking around the body sensor being set to automatically below the head but without the rotational piece but that would prevent your body from rotating from the origin. I was just worried about  the mounting onto the headset. I could use edtracker to track my body but then would need to make the modifications to the driver myself and not sure if it would be worth it just for me.

sigboe, 2016-06-11:
@betavr thankyou :)

Bletotum, 2016-07-04:
Yeah this would be great. It would be super goofy looking but it would function a lot more accurately!

sigboe, 2016-09-09:
Any news on this?

betavr, 2018-08-13:
This is on hold. Perhaps after making the driver functional again it can be reopened.

closed 2018-08-13 by betavr