betavr/steamvr_driver_hydra #4: Testing velocity and acceleration extrapolation.

Relys, 2016-04-22:
I spent a couple of hours messing around with extrapolation. Maybe you can get the values right? I was having issues with damping so I disabled the checks. I think the issue is that negative acceleration and velocity values represent the other side of an coordinate in 3D space so we need to account for that. As for accuracy maybe it would help to take a larger sample to calculate where the velocity and acceleration values converge? The sixense library has sixenseMath::Vector3 getVelocity(); sixenseMath::Vector3 getAcceleration(); functions which take current derivative but nothing for rotational and angular momentum so I am not using them. 

I also looked at the Unreal Hydra plugin for reference: https://github.com/getnamo/hydra-ue4/blob/master/Source/HydraPlugin/Private/FHydraPlugin.cpp#L441

Anyways, here are my changes:
https://github.com/Relys/driver_hydra/commit/a462252a2f1528b169883e67a3b135279f18f396

betavr, 2016-04-22:
Thank you, will test it as soon as I have free time. In the OSVR based driver the jitter was quite bad when I added acceleration data. According to the comments in Valve's code, this is expected with this driver too:

```
	// Note that using first-order derivatives of position is a terrible
	// way to actually supply velocity to the driver.  Good prediction is
	// required to compensate for unavoidable system latencies, and having
	// a good velocity estimate is key to that.  Any serious driver should
	// be doing sensor fusion with an IMU, and should have some decent
	// notion of instantaneous velocity.
	// (Also I have no idea what shenanigans are going on in this sixense_utils
	// class, so I hope it's reasonable)
```

```
	// The tradeoff here is that setting a valid velocity causes the controllers
	// to jitter, but the controllers feel much more "alive" and lighter.
	// The jitter while stationary is more annoying than the laggy feeling caused
	// by disabling velocity (which effectively disables prediction for rendering).
	// Even the Hydra (without IMU) could probably produce a better velocity here
	// with a different filter on top of the raw position.  Perhaps someone feels
	// like writing one??

```

Relys, 2016-04-27:
So, I've been thinking about this a little more and we should be doing linear extrapolation as opposed to just calculating the first order derivative: https://en.wikipedia.org/wiki/Extrapolation I think we can model this more accurately. We also need to figure out good dampening values (I really need to go in and put some debug print statements).

betavr, 2016-04-27:
It won't be easy to feed the prediction engine with correct values. I'm thinking of using the 3 unused buttons on the hydra to enable ingame finetuning of the parameters, so we can find a usable setting more easily.

Also merged your changes in the f/imu_emulation branch, and made it switchable from settings.

Relys, 2016-05-02:
So using driver from the latest installer has an issue. Pressing button 1 completely disables the controller (it just vanishes) as opposed to toggling velocity and acceleration. Also, the official sixense velocity and acceleration functions do not support angular velocity and acceleration so you can't put spins on things (like planets in the solar system demo).

3Duke, 2016-05-02:
Yes, it's true that Relys says: when I push button 1 the controller is hidden, and you can't throw nothing; it's possible to emulate angular velocity in some form?

betavr, 2016-05-02:
Disappering controllers are under investigation, and will be fixed soon.

We can add any kind of faked IMU data, but it is a very hard to test and finetune. So I'm doing the testing in Unity first, becase it's much easier to change the parameters in real time.

betavr, 2016-05-03:
Disappearing wands fixed: https://github.com/betavr/steamvr_driver_hydra/releases/tag/v1.0.1.4

@Relys Yes, this version only forwards the velocity and acceleration data from the sixense sdk. So no angular momentum is transferred currently, but I'm going to address this in the next update.

betavr, 2016-05-06:
First attempt at angular velocity calculation is released: https://github.com/betavr/steamvr_driver_hydra/releases/tag/v1.0.1.5

3Duke, 2016-05-08:
driver 1.0.1.5 is working ok, I can throw  sticks in The Lab now ! There is some jitter although, but I'm sure you'll do your magic and in the next version it'll be corrected (perhaps doing some filtering ?)
Lot of thanks Betavr !

betavr, 2016-05-11:
v1.0.1.7 is out, with reduced jitter and fixed angular velocity axis.

JonathanVB, 2016-05-17:
I get good results suppressing jitter just with exponential smoothing, i.e.

```
lastVel[0] = expFactor*vel[0] + (1 - expFactor)*lastVel[0];
lastVel[1] = expFactor*vel[1] + (1 - expFactor)*lastVel[1];
lastVel[2] = expFactor*vel[2] + (1 - expFactor)*lastVel[2];
m_Pose.vecVelocity[0] = lastVel[0];
m_Pose.vecVelocity[1] = lastVel[1];
m_Pose.vecVelocity[2] = lastVel[2];
```

for both the velocity and for the angular velocity using a factor of around 0.1.

betavr, 2016-05-17:
Awesome, will give it a try, thanks!

1337gmr, 2016-05-18:
With 1.0.1.7, I should be able to throw sticks in The Lab? Is there anything I can check to make sure I didn't screw anything up, I may be loading a bad driver.

betavr, 2016-05-18:
You need to press BUTTON 1 to activate the IMU emulation feature.

betavr, 2016-05-20:
@JonathanVB I added your smoothing method in a new release: https://github.com/betavr/steamvr_driver_hydra/releases/tag/v1.0.1.8

closed 2018-08-13 by betavr