betavr/steamvr_driver_hydra #5: Mapping Razer Hydra to Vive Wands

TechnoHunter, 2016-04-24:
It seems when using Steam, I am having problems with starting some of the demos or moving around with lack of buttons controls that are provided in the Vive controllers. Do you have a suggestion or fix for that? 

Thanks!

betavr, 2016-04-24:
All of the Vive's buttons are mapped to the Hydra, check out the button mapping table: https://github.com/betavr/steamvr_driver_hydra/wiki

TechnoHunter, 2016-04-25:
So the "Analog trigger, dual action emulated near the bottom of the travel" is how to teleport in the SteamVR apps?

betavr, 2016-04-25:
In most games you can teleport by pressing the joystick button. It really depends on the configuration of that particular app - in some games you can't teleport at all (like the Robot Repair demo).

closed 2016-06-18 by betavr