betavr/steamvr_driver_hydra #6: Steam VR not detecting it at all

harryjt, 2016-04-24:
Steam VR isn't detecting my hydras, whatever I do it wont work

TechnoHunter, 2016-04-24:
Have you tried copying the drivers in both Steam\Config and %ProgramFiles(x86)%\Steam\steamapps\common\SteamVR\drivers%? Also making sure to have the steamvr.vrsettings in the steamvr folder as well?

betavr, 2016-04-24:
What is the path to your SteamVR install folder?

djdevin, 2016-04-26:
same here, but to answer the question instead of "SteamVR" mine has a path of "OpenVR", so the installer scripts won't work

harryjt, 2016-04-27:
For some reason I don't have a vrserver.txt https://gyazo.com/0dd3cf825f146a3c09203d466f573c33 and the path for my steam VR is on my C: drive

betavr, 2016-04-27:
@harryjt Please include the output of vrpathreg.exe (probably found in "c:\Program Files (x86)\Steam\steamapps\common\SteamVR\bin\win32" folder).

@djdevin I assume that install path was used in the past. Have you tried reinstalling SteamVR?

djdevin, 2016-04-27:
@betavr actually I installed very recently. have you tried reinstalling SteamVR? :)

For now I copied the driver folder into drivers instead of using vrpathreg, and confirmed the hydras were working in the compositor!

betavr, 2016-04-27:
@djdevin Alright, I'm going to modify the install script then. Can you post your full steamvr install path please?

rosts, 2016-04-28:
My vrpathreg.exe gives:

"C:\Program Files (x86)\Steam\steamapps\common\SteamVR\bin\win64\vrpathreg.exe"
Runtime path = C:\Program Files (x86)\Steam\steamapps\common\SteamVR\
Config path = G:\steam\config
Log path = G:\steam\logs
External Drivers:
        C:\Program Files (x86)\SteamVR Hydra driver\hydra"

No idea why the config/log path is in G:. I even reinstalled steam and didn't add G: as a folder and it's the same output, perhaps it didn't delete the registry files upon deletion?

cbuchner1, 2016-05-21:
You might want to check out the config_tool I added to my Leap Motion repository. It uses a C++ program and RapidJSON to configure the driver path and to amend the streamvr.vrconfig file with driver specific information. This means previous user settings are fully retained, yay.

The .msi installer had to be changed to target 64 bit machines and the config_tool.exe had to be included as a 64 bit version, otherwise I think it ended up reading registry entries from the WOW6432 registry Node.

I think this method of installation is more reliable with respect to differing locations of Steam and StreamVR paths.

betavr, 2016-05-21:
Thanks, will check it out!

betavr, 2016-06-18:
The driver is going to be downloadable from Steam soon, so I'm closing this.

closed 2016-06-18 by betavr