betavr/steamvr_driver_hydra #10: Pitch adjustment

WugWiZard, 2016-05-06:
Enhancement request:

Pointing to items low on the screen in SteamVR is very uncomfortable, it should be possible to adjust the "aiming" height of the controller so that it aims lower.

cplr, 2016-05-06:
Your height might be calibrated wrong in the SteamVR setup. If it were uncomfortable for you to point at the ground in real life, then this request would make more sense to me..

WugWiZard, 2016-05-06:
Ah, ok, the problem lies in using the hydra while sitting, pointing to the ground is not possible due to the desk.  This is then not a hydra problem, as it is shared by the vive controllers, the interface is simply not designed for sitting at desk use (the desk prevents "normal" pointing at the ground, necessitating a rather uncomfortable bending of the wrists to achieve this above the table).

closed 2016-05-06 by WugWiZard

betavr, 2016-05-06:
I also encountered this problem when sitting.. probably real wands have this problem too because I an see ppl complaining about this on the steam forums. I think probably Steam is going to allow changing the placement of the dashboard sooner or later.