betavr/steamvr_driver_hydra #11: Hydra not detected

arronhuk, 2016-05-12:
My hydra isn't being loaded as a motion controller in steam even though the lighthouse is finding it.

It was working as a mouse in the tutorial demo so maybe I need to set something up differently in the razer software?

[vrserver.txt](https://github.com/betavr/steamvr_driver_hydra/files/262207/vrserver.txt)

betavr, 2016-05-12:
Try to disable or uninstall any razer / sixense software. You don't need to install anything other than this driver for steamvr to work.

arronhuk, 2016-05-12:
Hi,
Thanks for the quick response, I have uninstalled the razer drivers, rebooted, removed the steam hydra drivers and then reinstalled the steam hydra driversd ands I'm still not getting the controllers recognised. 
[vrserver.txt](https://github.com/betavr/steamvr_driver_hydra/files/262261/vrserver.txt)

betavr, 2016-05-12:
Ok, so the problem is the install script couldn't detect your steamvr installation, because it isn't under the default c:\program files(x86)\steam\steamapps folder. To solve this you can:
a) move hydra folder from where you installer the driver to the steamvr\drivers folder.
b) run vrpathreg.exe adddriver <path where you installed the driver>, eg. vrpathreg.exe adddriver "C:\Program Files\SteamVR Hydra driver\hydra"

cplr, 2016-05-12:
does the install script detect if the install directory is titled OpenVR instead of SteamVR?

betavr, 2016-05-12:
Nope, I've yet to find a way to auto detect steamvr's path from the registry. Currently it is hard coded in the post install script: %SteamPath%\steamapps\common\SteamVR\

arronhuk, 2016-05-12:
thanks for the help in the end I used the command D:\SteamLibrary\SteamApps\common\SteamVR\bin\win64\vrpathreg.exe adddriver "C:\Program Files (x86)\SteamVR Hydra driver\hydra" to get it working and had a quick go of the lab until it crashed. although this is a steamVR issue on my machine and probably just needs a re-install.

Thanks for your help and the work you have put in on getting this working

betavr, 2016-05-12:
No problem, I'm glad you worked it out. The installer script definitely needs some improvement..

closed 2016-05-12 by betavr