betavr/steamvr_driver_hydra #12: SteamVR not seeing Hydras at all.

OverlordMIke, 2016-05-16:
Followed all instructions given. Uninstalled Steam from it's D: installation and reinstalled it in C:, still no luck. Tell me what you need and where it is, I'll provide.

betavr, 2016-05-16:
Try to copy the hydra folder from the driver install to the ...\SteamVR\drivers directory.

OverlordMIke, 2016-05-16:
I've tried that, still nothing. I also do have ""activateMultipleDrivers" : true,"

betavr, 2016-05-17:
That's weird. We're going to need the contents of your vrserver.txt (found in steam\logs), and also the output of vrpathreg.exe (in steamvr\bin) to further analyze the problem.

OverlordMIke, 2016-05-17:
There doesn't seem to be a vrserver.txt in the logs folder, and vrpathreg.exe just flashes a prompt window up and goes away instantly.

OverlordMIke, 2016-05-17:
I just started SteamVR and it said they were connected but not tracking motion. So I quit out and got my Rift DK2 on and ready, started SteamVR again, and they weren't connected again. Can't seem to reproduce what I just got...

OverlordMIke, 2016-05-17:
I've figured it out, and it's odd. Turns out that the logs and steamvr config are still being read from the location on D:\, so I placed everything in there and it is working fine. It's completely ignoring the files and folders on C:\ and defaulting the D:\.

closed 2016-05-17 by OverlordMIke

cbuchner1, 2016-05-17:
NOTE: the log and config paths can be adjusted using the vrpathreg tool as
well.

betavr, 2016-05-17:
@OverlordMIke I'm glad you worked it out. Vpathreg.exe is a command line tool, so you need to open a command line and try to run it from there. As @cbuchner1 pointed out the configuration values (log directory, etc) can only be altered by this tool.