betavr/steamvr_driver_hydra #13: several issues please help

brdsnow, 2016-05-18:
I am using v1.0.1.7 using the installer didn't work.  I checked the log file and no hydra driver was being loaded.  I then opened the install script and manually registered the directory and moved the config file over.  Still didn't work right.  The .dll wouldn't load... for some reason it wanted it named the same as the directory I had installed it to?  Finally, I got it to see the hydra but... when I put the head set on I do not get any prompt to setup the hydra controllers.  

I also tried to do the room scale setup and the system sees my Headset but not the controllers.  I tried clicking all kinds of buttons and cannot get it to do anything useful.  I do know that it is seeing the controllers somewhat because when I run the tutorial the "vive" controller is above me and the trigger moves in and out but no tracking.  

Any suggestions?

betavr, 2016-05-18:
Could you post your vrserver.txt contents?

1337gmr, 2016-05-18:
I'm having the same problem - here is my vrserver.txt contents:

Wed May 18 2016 00:11:22.133 - ================================================================================================
Wed May 18 2016 00:11:22.133 - ================================================================================================
Wed May 18 2016 00:11:22.133 - VR server (v1461626459) starting up with config=C:\Program Files (x86)\Steam\config
Wed May 18 2016 00:11:22.134 - Load Json Settings from C:\Program Files (x86)\Steam\config\steamvr.vrsettings
Wed May 18 2016 00:11:22.192 - New Connect message from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\tools\bin\win32\vrmonitor.exe 20872 
Wed May 18 2016 00:11:22.198 - Unable to load driver driver_hydra_v1 from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\driver_hydra_v1\bin\win32\driver_driver_hydra_v1.dll.
Wed May 18 2016 00:11:22.198 - Unable to load driver driver_hydra_v1 because of error 103. Skipping.
Wed May 18 2016 00:11:22.215 - lighthouse: Attached HID Devices:
Wed May 18 2016 00:11:22.215 - lighthouse:     2833, 31: s/n WMHD302G500QPD: Rift (max-input-report 1)
Wed May 18 2016 00:11:22.215 - lighthouse:     1532, 300: s/n \\?\hid#vid_1532&pid_0300&mi_01#8&234dcaa5&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}: Razer Hydra (max-input-report 1)
Wed May 18 2016 00:11:22.215 - lighthouse:     46D, C531: s/n \\?\hid#vid_046d&pid_c531&mi_01&col04#8&c94913a&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}: USB Receiver (max-input-report 1)
Wed May 18 2016 00:11:22.215 - lighthouse:     1532, 300: s/n \\?\hid#vid_1532&pid_0300&mi_00#8&367de6e3&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}: Razer Hydra (max-input-report 0)
Wed May 18 2016 00:11:22.215 - lighthouse:     46D, C531: s/n \\?\hid#vid_046d&pid_c531&mi_01&col05#8&c94913a&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}: USB Receiver (max-input-report 1)
Wed May 18 2016 00:11:22.215 - lighthouse:     5AC, 220: s/n \\?\hid#vid_05ac&pid_0220&mi_01#9&23c0596d&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}: Apple Keyboard (max-input-report 1)
Wed May 18 2016 00:11:22.215 - lighthouse:     46D, C531: s/n \\?\hid#vid_046d&pid_c531&mi_01&col02#8&c94913a&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}: USB Receiver (max-input-report 1)
Wed May 18 2016 00:11:22.215 - lighthouse:     2833, 330: s/n WMHD302G500QP: Rift Audio (max-input-report 3)
Wed May 18 2016 00:11:22.215 - lighthouse:     46D, C531: s/n \\?\hid#vid_046d&pid_c531&mi_01&col03#8&c94913a&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}: USB Receiver (max-input-report 1)
Wed May 18 2016 00:11:22.215 - lighthouse:     9886, 1: s/n \\?\hid#vid_9886&pid_0001&mi_03#8&17b80faf&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}: ASTRO Wireless Transmitter  (max-input-report 3)
Wed May 18 2016 00:11:22.215 - Driver lighthouse has no HMDs. Skipping
Wed May 18 2016 00:11:22.244 - oculus: [CAPI] LibOVR module is located at D:\Oculus\Support\oculus-runtime\LibOVRRT32_1.dllWed May 18 2016 00:11:22.258 - oculus: IAD changed to 64.0mmWed May 18 2016 00:11:22.308 - oculus: Initialized Oculus device WMHD302G500QPD
Wed May 18 2016 00:11:22.308 - PoseUpdated from driver 2 for object 0 which is actually from -1
Wed May 18 2016 00:11:22.308 - Rejecting duplicate pending tracked device serial number WMHD302G500QPD_Camera
Wed May 18 2016 00:11:22.308 - Unable to load driver driver_hydra_v1 from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\driver_hydra_v1\bin\win32\driver_driver_hydra_v1.dll.
Wed May 18 2016 00:11:22.340 - error VRInitError_Driver_RuntimeOutOfDate when initing driver oculus_legacy from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\oculus_legacy\bin\win32\driver_oculus_legacy.dll.
Wed May 18 2016 00:11:22.371 - error VRInitError_Init_HmdNotFound when initing driver null from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\null\bin\win32\driver_null.dll.
Wed May 18 2016 00:11:22.371 - Active HMD set to oculus.WMHD302G500QPD
Wed May 18 2016 00:11:22.371 - Processing message VRMsg_Connect from vrmonitor (20872) took 0.179 seconds
Wed May 18 2016 00:11:22.372 - DXGI thread failed to get SDL display index
Wed May 18 2016 00:11:22.574 - New Connect message from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\bin\win32\vrcompositor.exe 1100 
Wed May 18 2016 00:11:22.579 - Using existing HMD oculus.WMHD302G500QPD
Wed May 18 2016 00:11:22.579 - Processing message VRMsg_Connect from vrcompositor (1100) took 0.00515 seconds
Wed May 18 2016 00:11:23.825 - Processing message VRMsg_CreateSwapTextureSet from vrcompositor (1100) took 0.512 seconds
Wed May 18 2016 00:11:24.149 - Processing message VRMsg_AddApplicationManifest from vrmonitor (20872) took 0.00786 seconds
Wed May 18 2016 00:11:24.155 - Processing message VRMsg_StartProcess from vrmonitor (20872) took 0.0057 seconds
Wed May 18 2016 00:11:24.373 - New Connect message from C:\Program Files (x86)\Steam\Steam.exe 5916 
Wed May 18 2016 00:11:24.378 - Using existing HMD oculus.WMHD302G500QPD
Wed May 18 2016 00:11:24.378 - Processing message VRMsg_Connect from Steam (5916) took 0.00524 seconds
Wed May 18 2016 00:11:24.607 - New Connect message from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\tools\bin\win32\vrdashboard.exe 15876 
Wed May 18 2016 00:11:24.613 - Using existing HMD oculus.WMHD302G500QPD
Wed May 18 2016 00:11:24.613 - Processing message VRMsg_Connect from vrdashboard (15876) took 0.00541 seconds
[... cut ...]
Wed May 18 2016 00:14:35.724 - Using existing HMD oculus.WMHD302G500QPD
Wed May 18 2016 00:14:35.724 - Processing message VRMsg_Connect from Steam (5916) took 0.00654 seconds
Wed May 18 2016 00:14:35.987 - New Connect message from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\tools\bin\win32\vrdashboard.exe 19696 
Wed May 18 2016 00:14:35.992 - Using existing HMD oculus.WMHD302G500QPD
Wed May 18 2016 00:14:35.993 - Processing message VRMsg_Connect from vrdashboard (19696) took 0.00582 seconds

betavr, 2016-05-18:
Your problem is the invalid folder name in steamvr\drivers. The directory must be named "hydra" not "driver_hydra_v1".

brdsnow, 2016-05-18:
[vrserver.txt](https://github.com/betavr/steamvr_driver_hydra/files/269944/vrserver.txt)

betavr, 2016-05-18:
The driver is successfully loaded according to this log, but you have many invalid external drivers registered. First try to unregister those with "vrpathreg removedriver <path>".

1337gmr, 2016-05-18:
Thanks betavr, it looks like the installer was putting the files in the wrong folder.
This is the default from the installer.
C:\Program Files (x86)\SteamVR Hydra driver\

Anyone know about the momentum not working?

betavr, 2016-05-18:
That's actually the correct behaviour. The installer copies the files outside the steam install, and registers as an external driver with vrpathreg. If this operation fails for you can always put the driver folder manually under SteamVR\drivers.

Also you can register the driver by issuing the command:

vrpathreg.exe adddriver "C:\Program Files (x86)\SteamVR Hydra driver\"

1337gmr, 2016-05-18:
Thanks for the help @betavr  it worked once I moved it!

brdsnow, 2016-05-19:
ok fixed sort of the removedriver didn't work... but moving the driver into steamvr folder did... actually had 2 sets connected until i changed the dir to stop the copy for loading externally.  However, now I have a problem with Jitter. anything more than 2 feet from the base station shakes so much that you can't do anything?  Is there any filtering that can be done?  I've seen others use the razers without this issue.

betavr, 2016-05-19:
BUTTON 1 toggles the imu emulation, which causes lots of jitter. If you still get jitter with imu emulation off, then probably there is a some magnetic field interfering with your hydra.

kamtung, 2016-05-22:
Hi betavr, can you help to take a look what have I done wrong? I can't connect my Hydra to SteamVR

Sat May 21 2016 21:08:50.797 - ================================================================================================
Sat May 21 2016 21:08:50.797 - ================================================================================================
Sat May 21 2016 21:08:50.797 - VR server (v1461626459) starting up with config=C:\Program Files (x86)\Steam\config
Sat May 21 2016 21:08:50.798 - Load Json Settings from C:\Program Files (x86)\Steam\config\steamvr.vrsettings
Sat May 21 2016 21:08:50.798 - C:\Program Files (x86)\Steam\config\steamapps.vrmanifest - No applications in manifest
Sat May 21 2016 21:08:50.891 - New Connect message from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\tools\bin\win32\vrmonitor.exe 4688 
Sat May 21 2016 21:08:50.894 - Unable to load driver SteamVR Hydra driver from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\SteamVR Hydra driver\bin\win32\driver_SteamVR Hydra driver.dll.
Sat May 21 2016 21:08:50.894 - Unable to load driver SteamVR Hydra driver because of error 103. Skipping.
Sat May 21 2016 21:08:50.897 - Unable to load driver hydra from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\hydra\bin\win32\driver_hydra.dll.
Sat May 21 2016 21:08:50.897 - Unable to load driver hydra because of error 103. Skipping.
Sat May 21 2016 21:08:50.899 - Unable to load driver hydra from C:\Program Files (x86)\SteamVR Hydra driver\hydra\bin\win32\driver_hydra.dll.
Sat May 21 2016 21:08:50.899 - Unable to load driver hydra because of error 103. Skipping.
Sat May 21 2016 21:08:50.902 - Unable to load driver hydra from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\SteamVR Hydra driver\hydra\bin\win32\driver_hydra.dll.
Sat May 21 2016 21:08:50.902 - Unable to load driver hydra because of error 103. Skipping.
Sat May 21 2016 21:08:50.905 - Unable to load driver hydra from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\hydra\bin\win32\driver_hydra.dll.
Sat May 21 2016 21:08:50.905 - Unable to load driver hydra because of error 103. Skipping.
Sat May 21 2016 21:08:50.995 - lighthouse: Attached HID Devices:
Sat May 21 2016 21:08:50.995 - lighthouse:     46D, C52B: s/n \\?\hid#vid_046d&pid_c52b&mi_02&col01#8&2b7c0e04&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}: USB Receiver (max-input-report 2)
Sat May 21 2016 21:08:50.995 - lighthouse:     46D, C52B: s/n \\?\hid#vid_046d&pid_c52b&mi_02&col02#8&2b7c0e04&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}: USB Receiver (max-input-report 2)
Sat May 21 2016 21:08:50.995 - lighthouse:     46D, C52B: s/n \\?\hid#vid_046d&pid_c52b&mi_02&col03#8&2b7c0e04&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}: USB Receiver (max-input-report 2)
Sat May 21 2016 21:08:50.995 - lighthouse:     1532, 300: s/n \\?\hid#vid_1532&pid_0300&mi_01#8&3306968b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}: Razer Hydra (max-input-report 1)
Sat May 21 2016 21:08:50.995 - lighthouse:     46D, C52B: s/n \\?\hid#vid_046d&pid_c52b&mi_01&col02#8&13b881bf&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}: USB Receiver (max-input-report 1)
Sat May 21 2016 21:08:50.995 - lighthouse:     1532, 300: s/n \\?\hid#vid_1532&pid_0300&mi_00#8&31ec97a5&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}: Razer Hydra (max-input-report 0)
Sat May 21 2016 21:08:50.995 - lighthouse:     46D, C52B: s/n \\?\hid#vid_046d&pid_c52b&mi_01&col03#8&13b881bf&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}: USB Receiver (max-input-report 1)
Sat May 21 2016 21:08:50.995 - lighthouse:     46D, C52B: s/n \\?\hid#vid_046d&pid_c52b&mi_01&col04#8&13b881bf&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}: USB Receiver (max-input-report 1)
Sat May 21 2016 21:08:50.995 - Driver lighthouse has no HMDs. Skipping
Sat May 21 2016 21:08:50.999 - error VRInitError_Driver_RuntimeOutOfDate when initing driver oculus from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\oculus\bin\win32\driver_oculus.dll.
Sat May 21 2016 21:08:50.999 - Unable to load driver oculus because of error 204. Skipping.
Sat May 21 2016 21:08:51.001 - error VRInitError_Driver_RuntimeOutOfDate when initing driver oculus_legacy from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\oculus_legacy\bin\win32\driver_oculus_legacy.dll.
Sat May 21 2016 21:08:51.001 - Unable to load driver oculus_legacy because of error 204. Skipping.
Sat May 21 2016 21:08:51.022 - oculus_vridge: Could not get settings API. Using default values
Sat May 21 2016 21:08:51.022 - oculus_vridge: driver_vridge: Serial Number: 0.8
Sat May 21 2016 21:08:51.022 - oculus_vridge: driver_vridge: Model Number: Vridge
Sat May 21 2016 21:08:51.022 - oculus_vridge: driver_vridge: Window: -1 -1 2560 1440
Sat May 21 2016 21:08:51.022 - oculus_vridge: driver_vridge: Render Target: 1280 1440
Sat May 21 2016 21:08:51.022 - oculus_vridge: driver_vridge: Seconds from Vsync to Photons: 0.000000
Sat May 21 2016 21:08:51.022 - oculus_vridge: driver_vridge: Display Frequency: 58.000000
Sat May 21 2016 21:08:51.022 - oculus_vridge: driver_vridge: IPD: 0.065000
Sat May 21 2016 21:08:51.022 - Driver oculus_vridge has no HMDs. Skipping
Sat May 21 2016 21:08:51.023 - error VRInitError_Init_HmdNotFound when initing driver null from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\null\bin\win32\driver_null.dll.
Sat May 21 2016 21:08:51.023 - Unable to load driver null because of error 108. Skipping.
Sat May 21 2016 21:08:51.023 - Active HMD not found. Returning best error VRInitError_Init_HmdNotFound
Sat May 21 2016 21:08:51.023 - Allowing driver load failure VRInitError_Init_HmdNotFound for application vrmonitor (4688) because it is vrmonitor
Sat May 21 2016 21:08:51.023 - Processing message VRMsg_Connect from vrmonitor (4688) took 0.131 seconds

kamtung, 2016-05-22:
My windows is x64. SteamVR can detected the controller after installed "x86 version" (It doesn't work with x64 version)

The controller icon are flashing in stermVR. Show "Controller not tracking" .so working through this. at least I see green now :+1:

betavr, 2016-05-22:
Yes, the VC redist is needed for some ppl, I'm glad it solved the problem for you. The green flashing is normal until the hemisphere calibration is completed. You should see a message in the headset telling you to aim at the base station and pull both triggers. After that the flashing stops and you can calibrate by pressing the system buttons.

closed 2016-06-18 by betavr

iBringTheStorm, 2018-01-26:
Guys I need some serious help. I just recieved my Hydra yesterday and installed the razer configurator and proceeded to steam to install SteamVR ( I also downloaded the Sixense SDK but have no clue how to instal it or use it whatsoever) and the Razer Hydra driver from Steam, the driver from steam was not detected and/or installed properly. So I uninstalled steam and went through the various other fixes and nothing seems to work. I don't know anything about programming and I could really use some assistance.

iBringTheStorm, 2018-01-26:
Also the executable file for hydra driver 1.1.2 download link is broken @betavr