betavr/steamvr_driver_hydra #14: Sometimes one of the controllers gets turned upside down.. doesnt reset properly?

vonkoda, 2016-05-22:
Is there a more.. intensive or reliable reset mechanism than the Menu + Shoulders method? Occasionally but TBH eventually in most games I tried one of the two controllers seems to just flip out and get set upside. Reseting doesnt put it right but does move it around.

Is it a known problem?

betavr, 2016-05-22:
This is probably caused by accidental triggering of the hemisphere calibration. This should only happen when you place the controllers on the base station, but sometimes it is triggered when holding them both in close proximity to it.

First I'd try moving the base station a little farther to make this less likely to happen. 

It should be possible to somehow completely reset the tracking, gonna look into it.

vonkoda, 2016-05-22:
Cool I'll give it a shot. The awkwardness on the Hydra (vs I assume how a Vive works) is that you need to put it mid way between the ground and your head and it seems to want to be at the edge of the desk. If you dont do that you cant pick things up in VR, it loses tracking near the ground!

closed 2018-08-13 by betavr