betavr/steamvr_driver_hydra #15: Missing

koganei, 2016-05-26:


betavr, 2016-05-26:
So the game needs you to place your thumb on the center of the touchpad without pressing it?

koganei, 2016-05-26:
Yeah, exactly.

nintenjo, 2016-06-18:
Another issue related to this is some SteamVR games require you to simply place your thumb on the tracker (like a touch pad) while others require you to actually depress the tracker like a button.. Doing this on an analogue stick is very clunky :( 

(I know this is more related to the way the game input options have been written, but is a frustration. A good example is Windlands that requires you to click an hold the analogue sticks to move / jump making it virtually unplayable.

bobjwatts, 2016-06-19:
Would love to see a solution, The Gallery has this issue.

betavr, 2016-06-25:
The feature has been Implemented in the latest release: https://github.com/betavr/steamvr_driver_hydra/releases/tag/v1.0.1.12

bobjwatts, 2016-06-25:
Thanks betavr. I can see the the thumbpad dot lights up when I press '3' but I'm not sure it's really 'hold the thumbpad' rather it's 'touch the thumbpad'. In Call of the Starseed you have to 'press' the thumbpad to use blink, when I press '3' it doesn't register. Sorry I don't have a Vive to test.

betavr, 2016-06-25:
Unfortunately I don't have that game, so I can't test it either. Have you tried pressing button 3 and the joystick button at the same time?

bobjwatts, 2016-06-26:
I just gave it another go, trying everything I could, no luck.

AlexLogin0ff, 2016-06-26:
In Call of the Starseed is not working thumbpad :(

forrestv, 2016-07-05:
I can confirm that this still isn't working. Here's a screenshot of the prompt in the game:

[image]

@betavr, any idea what could be missing? I can try things (recompiling, etc), but it looks like the driver is doing all it can... What could Starseed be looking at besides the `k_EButton_Axis0` bit in `ulButtonTouched`/`ulButtonPressed`?

forrestv, 2016-07-05:
Hypotheses I have:

* The k_EButton_Button3 pressed/touched events somehow result in Starseed ignoring the k_EButton_Axis0 touch event
* Starseed needs TouchPadMove events, which could possibly be injected via the VendorSpecificEvent callback

betavr, 2016-07-05:
Thanks for the investigation. Will upload a new version soon without the Button3 keypress event.

Btw does the reticle appear in the game (step 2. on the picture)?

betavr, 2016-07-05:
Another quick fix: the joystick axis is now reported as zero when the deadzone is active, please try it: https://github.com/betavr/steamvr_driver_hydra/releases/tag/v1.0.1.14

Looking forward to what you find with touchpadmove!

forrestv, 2016-07-05:
That didn't help either. :/

betavr, 2016-07-05:
Nvm, I contacted the developers. Will report back when they respond.

chedigitalproduction, 2016-07-05:
The developers have stated that they won't support Hydra, even though it was a stretch goal in the Kickstarter. Hopefully they do respond though, looks like we are getting closer to understanding the issue.

betavr, 2016-07-06:
Yeah, received a negative response. To sum it up, they are satisfied with the current situation. They don't want Oculus users to play the game until it has been officially launched for Oculus platform: http://steamcommunity.com/app/270130/discussions/0/351660338732546659/

They are concerned that ppl aren't going to have the "proper experience".

betavr, 2016-07-07:
Ok, I just added a new section in the wiki: https://github.com/betavr/steamvr_driver_hydra/wiki#list-of-unsupported-games

I think this case is closed.

closed 2016-07-07 by betavr

bobjwatts, 2016-07-07:
Sad face. Thanks for trying betavr.

betavr, 2016-07-07:
Yeah kinda disappointing. Oculus is trying hard to build a fence around it's garden, and I wouldn't be too surprised to find out that they have an agreement with the Cloudhead guys.

forrestv, 2016-07-09:
It appears that this problem isn't fixable from within steamvr_driver_hydra. It's a result of directly checking whether the *HMD* is a Vive.

On the bright side, you can work around this by pressing W to blink and using the mouse's scroll wheel to rotate the play area, although that isn't fun at all. Thanks, Cloudhead.

AlexLogin0ff, 2016-07-09:
this function can be specifically applied to the new individual driver для этой игры ?

betavr, 2016-07-09:
@forrestv So you're saying it is possible to blink using the keyboard and mouse? This is interesting, can you please elaborate?

@AlexLogin0ff Driver features can be enabled/disabled in the steamvr.vrsettings file, so there is no need for an individual driver.

nintenjo, 2016-07-09:
Is this not the same issue i reported above? I feel like its a separate issue now :) (in which case i should open another thread).

I had issue with the main SteamVR gallery and games that require you to click the touchpad directions as if it were a digital D-Pad.. this on the Hydra analog stick is very difficult to do as you have to push the direction and then click the stick in.. (simple in theory) but games like Windlands require you to use this method to navigate in first person.. I was looking for a solution that allowed users to toggle the function of the analog stick from default (as it is now) to a digital mode so any direction pressed on the stick also triggers a 'on-click' event on the touchpad? (perhaps button 4 toggles modes?)

betavr, 2016-07-09:
@nintenjo yeah, please open a separate issue for that one, thanks!