betavr/steamvr_driver_hydra #18: latest steam beta prevents initial calibration

djdevin, 2016-06-18:
after updating to the latest steam beta, it's impossible to get past the "point the hydra at the base station and press any button" prompt

the controllers still work, if you raise them to your head and press the system buttons

but the green box will not go away

reverting to an earlier steam beta works

betavr, 2016-06-18:
Thank you for the report. Probably related to this change:

> Enabled preliminary support for stereoscopic screenshots in SteamVR. They are taken by holding down the System button and pulling the Trigger on the SteamVR controller. Screenshots can be viewed in the SteamVR dashboard by selecting the application and choosing the screenshots panel. Sharing screenshots by uploading them Steam Community currently only shares the normal 2D image. Sharing of stereoscopic images will be forthcoming.

betavr, 2016-06-18:
Another problem: the platform name has been renamed from x64 to Win64 in the newest beta:

`Sun Jun 19 2016 00:48:22.043 - Unable to load driver hydra from d:\SteamLibrary\SteamApps\common\OpenVR Driver for Razer Hydra\hydra\bin\win64\driver_hydra.dll.`

djdevin, 2016-06-19:
Haven't had an issue with that one, but the releases notes for SteamVR also state that.

As a workaround for the calibration overlay, I just deleted the images in the /overlays/ folder and it works just fine.

betavr, 2016-06-19:
So, after debugging this for a few hours it is apparent that the problem isn't related to that screenshot stuff above. 

The root of the problem is corruption of the m_eHemisphereTrackingState enum. Using a simple integer member variable instead to track the state completely fixes the issue.

However I can see other very weird things:
- the controllers are visible even before calibration
- the calibration overlay flashes occasionally
- left and right controllers are switched up

At this point I'm pretty sure this is a bug in SteamVR. I can only recommend reverting to a previous SteamVR build until this is resolved.

betavr, 2016-06-19:
@djdevin as a temporary workaround create an environment variable "VR_DISABLE_64BIT_SERVER".  The 32 bit version of the driver apparently doesn't have any issues.

betavr, 2016-06-21:
@djdevin The problem has been fixed in the latest release: https://github.com/betavr/steamvr_driver_hydra/releases/tag/v1.0.1.10

For reference, it was the x64 sixense utils dll corrupting the contents of the drivers memory.

closed 2016-06-21 by betavr