betavr/steamvr_driver_hydra #21: SteamVR not detecting Razer Hydra

jefreestyles, 2016-06-26:
I tried installing the drivers and the room setup application won't detect the controllers.  I tried the setup executable in the steamvr_hydra_driver_v1.0.1.2.zip file and then I tried steamvr_hydra_driver_v1.0.1.12.msi.

I know the Hydra works because the base is lighting up and I can control the mouse cursor with the right stick when I press the middle button.

I have my steam installed in E:\Games\Steam.

jefreestyles, 2016-06-26:
Also, my SteamVR folder in my Steam folder is called OpenVR.  Is that weird?
E:\Games\Steam\steamapps\common\OpenVR

Maybe I need to uninstall and re-install SteamVR.

betavr, 2016-06-26:
Currently custom OpenVR installations aren't supported by the installer script. This means you need to manually register the driver in Steamvr. Please run vrpathreg.exe under your OpenVR\bin\win64 directory like this:

`vrpathreg adddriver "c:\Program Files (x86)\OpenVR Driver for Razer Hydra\hydra"`

Of course you are going to have to edit the path in my example to fit your needs (and make sure to keep the quotes). After adding the path, run vrpathreg.exe without parameters to have it list the registered drivers. In case you notice any other paths, you can remove them with `vrpathreg removedriver "invalid_path"`. Also check your vrserver.txt under Steam\logs to see any errors you get.

jefreestyles, 2016-06-26:
Thanks for your reply.  I'll look into your steps but I got it to work by deleting the SteamVR app from within the client, renaming the OpenVR folder to OpenVR_unused, re-installing SteamVR, and then deleting the old OpenVR folder.  I guess the old beta version of SteamVR named the folder OpenVR for some reason.

Now the only problem is I don't have enough space to play Vanishing Realms.  It requires a 2 x 2 space and I can only get up to 3 x 1.5 and that's with calibrating my Hydra over my desk and monitor.  I'll keep trying.

Keep up the great work!

betavr, 2016-06-26:
Yeah, room setup can be tricky. I'd try putting the base station on the floor in the center of the play area.

Ibsn87, 2016-06-29:
To anyone having any issues getting the GREEN CONTROLLER ICON in steam VR. 

I installed this package to default directory, This did not work, i had no green controllers in steam. I located the new files as per the install directory and copied those installed files into my custom steam directory as follows; 

Find your steam folder steam/steamapps/common/openvr/drivers and copy those files from the installer into here. 

Its amazing! CANT WAIT for the Touch controllers!

I also had to do the same thing for LEAP MOTION drivers from github in case anyone is wondering.

I hope this helps, im not a guru at any of this stuff!

betavr, 2016-06-29:
Yes, this is a known problem with the installer. You need reinstall SteamVR to have the install script detect it correctly (its latest path doesn't contain "OpenVR"). Also make sure to install it in the default steamapps directory (the one under the Steam directory).

**Please note these installation problems won't be fixed. The driver is going to be available directly from Steam soon, so this installer will be superseded, with Steam doing the installation and driver registration tasks.**

space123321, 2016-07-08:
I am really at a loss here - first thank you so much I had it working perfectly!  Yesterday, I plugged in my hydras and windows installed drivers for it (which I hadn't recently noticed).  Now SteamVR will not see my hydras - try uninstalling betavr and reinstalling, tried uninstalling steamvr and reinstalling, adding the files as per above and simply cannot get the controllers to register?

betavr, 2016-07-08:
If it's a windows driver problem then you can try to uninstall it from the device manager, then restart (without unplugging the hydra). Otherwise check vrserver.txt for anything related, it's in steam\logs.

space123321, 2016-07-08:
Thank you for the response - I can't seem to uninstall a driver via device manager?   This is the vrsrever hydra specific details I am receiving:

Fri Jul 08 2016 19:30:10.522 - Load Json Settings from C:\Program Files (x86)\Steam\config\steamvr.vrsettings

Fri Jul 08 2016 19:30:10.522 - Skipping external driver C:\Program Files (x86)\SteamVR Hydra driver\hydra because it is not a directory

Fri Jul 08 2016 19:30:10.599 - New Connect message from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\tools\bin\win32\vrmonitor.exe (VRApplication_VRMonitor) 8388 

Fri Jul 08 2016 19:30:12.634 -   driver hydra implements interfaces IVRSettings_001 ITrackedDeviceServerDriver_004 IVRDisplayComponent_002 IVRDriverDirectModeComponent_001 IVRControllerComponent_001 IVRCameraComponent_001 IServerTrackedDeviceProvider_003 IClientTrackedDeviceProvider_003 

Fri Jul 08 2016 19:30:12.634 - Loaded server driver hydra (IServerTrackedDeviceProvider_003) from D:\Program Files (x86)\SteamVR Hydra driver\hydra\bin\win64\driver_hydra.dll

Fri Jul 08 2016 19:30:12.634 - Driver hydra has no devices. Skipping

betavr, 2016-07-09:
Yeah, it looks like the steamvr driver is being loaded, but it can't find your hydra.

You can right click a device in device manager and select uninstall. If you can't identify which device it is, open up Control Panel > Device and Printers, right click the hydra, select properties, then you can view all the related usb devices on the Hardware tab.

space123321, 2016-07-10:
Thanks again - got the hydra removed from the device manager however still no luck - driver loads however does not find hydra... same skipped message.  So disappointed as I purchased several Vive games as the experience with the Hydra is so much more of a complete experience for the Rift.  Hopefully the direct driver from Steam you mention is released sooner than later and it am up and running again soon.

betavr, 2016-07-10:
Well the Steam distribution method won't help with this issue. Do you have motion creator, or any other sixense/razer software installed?

space123321, 2016-07-10:
Nothing else from Razor installed...  and  as mentioned everything was working great and it just stopped working as I believe a windows driver is overriding the betavr driver - so weird!  I am trying to locate the msiexec info to delete it manually without any luck.

betavr, 2016-07-10:
The windows driver is required for using the hydra for anything in windows, SteamVR included.

You can verify that your hydra is working in windows with the Sixense simple3d app. You can get it in the sdk (under the \samples\win64\ folder): http://sixense.com/windowssdkdownload

space123321, 2016-07-10:
Thanks - the app works.  I was able to get the hydras running again - thank you again for your help. After confirming that the hydras woked in the simple3d app - a clean install of steam and drivers worked.

betavr, 2016-07-10:
Awesome!

closed 2018-08-13 by betavr