betavr/steamvr_driver_hydra #22: Oculus software overriding SteamVR

Bletotum, 2016-07-03:
Randomly before five minutes into anything on SteamVR/OpenVR, my Rift goes to the loading screen with the black void and a white hourglass. There seems to be no solution but to completely exit the SteamVR game and relaunch it.

betavr, 2016-07-03:
Yeah I get this too sometimes, but not too often. Can I ask what is your configuration (cpu/gpu/windows version/etc)?

Bletotum, 2016-07-04:
Windows 8.1 64 bit. GTX 970. On my phone now but iirc I have an intel i5-4690k, if not then similar above/at recommended specs by Oculus.

I'm not sure if this is a steamvr issue specifically or if it's because of the hydras.

Bletotum, 2016-07-04:
Whenever the bug does happen though, steamvr says the headset is unresponsive in its little status window. Restarting steamvr seems to be the only solution. The games do not crash.

betavr, 2016-07-04:
Did you opt in the steamvr beta?

Bletotum, 2016-07-04:
I'm in the Steam beta branch, yes. The SteamVR status window titles itself as a beta.

betavr, 2016-07-04:
I'm not sure I can help with this. Maybe it's worth a try to open an issue at Valve: https://github.com/ValveSoftware/openvr/issues

Bletotum, 2016-07-04:
Well, this doesn't happen to me if I use BigScreen on SteamVR, come to
think of it.

rblostwolf, 2016-07-16:
so do you put steam into big picture mode then launch steam vr? been looking for a while for a solution for this and not found anything yet i have win 10 970 and an i5 6600 so maybe a 970 issue?

rblostwolf, 2016-07-16:
just tried doing that and it didn't work got the hourglass screan not long after starting the archery game in the lab

rblostwolf, 2016-07-16:
think i misunderstood you just meant you didnt have the issue with the bigscreen beta? ive not had it with space pirate trainer

Bletotum, 2016-07-17:
yeah the lab is really really bad with this bug

rblostwolf, 2016-07-29:
any solution for this yet?

monkeydaz, 2016-09-04:
I'm currently seeing this on 2 machines both Windows 10 64 bit. GTX 970 with touch controllers not Hydra's. The issue is really bad on the Lab 10-15 minutes in black screen with small white egg timer in bottom right. This issue is also present in"Job Simulator" and "Waltz of the wizard". On Waltz if I spam the steam menu overlay button the screen comes back on I continue playing normally (Until it happens again later) however this doesn't seem to work on the other titles.

I've tried leaving the rift on booted the game up and just put a piece of card over the rift "on-head" sensor thinking it may be the headset was somehow going to sleep. However the blackness issue didn't seem to happen even after 20-30 minutes of sitting on the game screen.

SixMarbles, 2016-09-08:
Seeing this issue as well, particularly in titles with more frequent frame drops. I have correlated this issue with high CPU load. Is it possible the issue is triggered by a conflict between the Vive's reprojection and Oculus runtime?

nVidia GeForce GTX 1080
AMD FX-4350 Quad-Core @ 4.2 GHz
16.0 GB DDR4
Asus M5A97 r2.0
Windows 10 Home x64

betavr, 2016-09-08:
Not sure if the high cpu load is the cause or an effect, because having high load alone doesn't trigger this bug. In my experience it pretty much comes randomly.

westingtyler, 2016-10-21:
This happens to me, too. Windows 10 Home x64, Steam package version 1277000231
I have an Oculus Rift CV1 and Oculus Touch controls.

Games that go to a black hourglass screen after a few minutes, with the game still playable by looking at the computer screen:

Hot Dogs, Horseshoes, and Hand Grenades
Windlands
Job Simulator
The Lab
Tilt Brush

The bug does NOT happen in Universe Sandbox 2 as far as I can tell.

closed 2018-08-13 by betavr