betavr/steamvr_driver_hydra #23: Trouble building project

Bletotum, 2016-07-07:
http://i.imgur.com/onLtxp2.png

I've linked the appropriate old version of OpenVR as instructed, as well as the steam-installed sixense SDK. I don't know what I'm supposed to do with this error, though, to try modding this.

betavr, 2016-07-07:
Yeah, you need to build it with the latest OpenVR sdk, and keep in mind the project is for VS2015 not 13.

Also make sure to download the sixense sdk from sixense, not from steam like the readme says! The steam version only has dlls for VS2010.

Bletotum, 2016-07-07:
I just tried it the way you said, with the latest OpenVR and the proper sixense under VS2015. I've still got the same error. Is there something I forgot?

betavr, 2016-07-07:
Ok, after taking a closer look at your pic, it says it's missing some overrides. So you probably have an old revision of the driver source.

Bletotum, 2016-07-07:
It's your current version of this fork with the other up to date sources.

betavr, 2016-07-08:
Ok, check your copy of driver_hydra.h, and make sure you can see these lines:

``virtual vr::ITrackedDeviceServerDriver * GetTrackedDeviceDriver( uint32_t unWhich ) override;``
``virtual vr::ITrackedDeviceServerDriver * FindTrackedDeviceDriver(const char * pchId) override;``

Bletotum, 2016-07-08:
Both of those lines are there. Later, in the same file, this: http://i.imgur.com/l5EQlqM.png

betavr, 2016-07-08:
I just checked and Openvr has been updated to 1.0.2, but the driver uses 1.0.0. It is possible this is the cause of your problem. Try to check out release 1.0.0 of openvr: https://github.com/ValveSoftware/openvr/releases/tag/v1.0.0

Bletotum, 2016-07-08:
That did the trick, thanks!

closed 2016-07-08 by Bletotum

betavr, 2016-07-09:
Well, I guess it's time for an openvr upgrade once again. Good to know it worked out for you!

betavr, 2016-07-09:
updated to OpenVR 1.0.2: https://github.com/betavr/steamvr_driver_hydra/commit/52f25c5dca261c8eea96f518e9f24f0fcb49f6ac