betavr/steamvr_driver_hydra #24: Analog Stick > Digital Controller Toggle.

nintenjo, 2016-07-10:
I had issue with the main SteamVR gallery and games that require you to click the touchpad directions as if it were a digital D-Pad.. this on the Hydra analog stick is very difficult to do as you have to push the direction and then click the stick in.. (simple in theory) but games like Windlands require you to use this method to navigate in first person (as if WSAD on a keyboard).. I was looking for a solution that allowed users to toggle the function of the analog stick from default (as it is now) to a digital mode so any direction pressed on the stick also triggers a 'on-click' event on the touchpad? (perhaps button 4 toggles modes?)

This issue will also affect a lot of games that support the Oculus Move controllers which have the same analog sticks as the Hydra.

betavr, 2016-07-29:
@nintenjo In the latest release I fixed a trackpad related problem. Can you please test how this change affects the Gallery?

nintenjo, 2016-08-08:
@betavr I have downloaded the latest version via Steam and the issue still has not resolved. 

For a repro case example you can:
- Load Steam VR
- Press the Home Button (on Hydra) to load the main VR Interface
- Try to use the Right analogue stick to navigate. (without clicking it).

closed 2018-08-13 by betavr