betavr/steamvr_driver_hydra #25: Razer Hydra not detected in some games

AlexLogin0ff, 2016-07-14:
Razer Hydra not detected in games (UE4) - NVIDIA® VR Funhouse, The Nest, Out of Ammo, Lightblade VR... :(

SooSiaal, 2016-07-14:
You can add raw data to the list,it detects controllers but not working right at all

betavr, 2016-07-14:
Thanks, they will be added to the unsupported list. 

Can you tell me in more detail why these games don't work?

betavr, 2016-07-15:
Also, can you please verify the method described in the post below works?
https://www.reddit.com/r/oculus/comments/4svgi0/nvidia_vr_funhouse_with_leap_motion/

SooSiaal, 2016-07-15:
for Raw Data,turning compatibility on or off doesnt change a thing, it sort of recognizes the controllers,but the hands ingame dont move along with the hydras (rotation does work somewhat),left trigger activates right trigger as well,those sort of things

AlexLogin0ff, 2016-07-15:
Do not work at the program level, you need to look  NVIDIA® VR Funhouse, free to Steam - http://store.steampowered.com/app/468700/

betavr, 2016-07-15:
@SooSiaal thanks!

@AlexLogin0ff I'd try it myself, but I can't launch it on my AMD gpu.

AlexLogin0ff, 2016-07-15:
the problem is solved, it will be interesting to all -
Rift + Hydras (compatibility mode for win 7 on VRFunhouse.exe) it works flawless and other games on UE4 !!!
Remains an open question with "Raw Data", controllers at the start of the game is not in his place and they can not be calibrated

rblostwolf, 2016-07-15:
not detecting them in fruit ninja vr

rblostwolf, 2016-07-15:
oh sorry just saw the compatibility thing you mentioned and it works now thanks for that

AlexLogin0ff, 2016-07-15:
tested -  NVIDIA® VR Funhouse, Гнездо, патроны, The Path of Greatest Resistance

AlexLogin0ff, 2016-07-15:
Tested! - NVIDIA® VR Funhouse, The Nest, Out of Ammo, PoolNation VR, The Path of Greatest Resistance

betavr, 2016-07-15:
Awesome news! :+1: Method added to the wiki, thank you everyone for testing!

Regarding Raw Data, I'll try to speak with the developer.

SooSiaal, 2016-07-15:
I asked them a qustion about it on facebook,think it got misunderstood as they said Hydra won't be supported :p
But as this driver emulates the vive wands i'm sure there are some possibilities getting it to work.. i'm no programmer so i can't be of much help there unfortunately

AlexLogin0ff, 2016-07-15:
it is possible that in this subject will tell you to fix driver - https://www.reddit.com/r/Vive/comments/4syoio/psa_ue4_assigns_the_first_controller_you_turn_on/?st=iqo7j0po&sh=1b3002c1

betavr, 2016-07-17:
Interesting, some ppl on reddit are saying that they can play Raw Data with the hydras:
https://www.reddit.com/r/oculus/comments/4sy8dv/raw_data_doesnt_work_with_rift_touch/d5df46h

AlexLogin0ff, 2016-07-17:
You can play the Raw Data, if you put a Hydra base at shoulder level in front of you

betavr, 2016-07-17:
Weird, so it basically ignores the calibration of the base station?

AlexLogin0ff, 2016-07-17:
In Raw Data the controllers by themselves, emulator only gives their inclusion in the game

SooSiaal, 2016-07-17:
That is odd,will test this

betavr, 2016-07-17:
@AlexLogin0ff maybe the game has built-in Hydra support, but it's broken. Can you try what happens if you press the system button (does the SteamVR dashboard appear)? Are the controllers still uncalibrated on the dashboard too?

AlexLogin0ff, 2016-07-17:
if you click on SteamVR menu, you can see emulated controllers, they are here just for its perfect location, located in the game one's own way

betavr, 2016-07-17:
Very strange, I can only assume it's accessing the hydra directly. Could you check if the game's install folder contains the file sixense.dll, or sixense_x64.dll?

AlexLogin0ff, 2016-07-17:
[image]

AlexLogin0ff, 2016-07-17:
[image]

betavr, 2016-07-17:
Wow, so they do have their own controller implementations. I'd try to remove the Sixense and SixenseStem folders, and see what happens in the game.

AlexLogin0ff, 2016-07-17:
I tried everything too!

betavr, 2016-07-17:
Delete all of the sixense_*.dll files from the game's folder, and make sure Steam doesn't reinstall them when you launch the game. That should disable their built-in hydra driver.

AlexLogin0ff, 2016-07-17:
version here - OpenVR v0_9_12

AlexLogin0ff, 2016-07-17:
Я буду стараться, я напишу позже

AlexLogin0ff, 2016-07-17:
unfortunately it did not help (

AlexLogin0ff, 2016-07-17:
may need to change in this file -
[image]

AlexLogin0ff, 2016-07-17:
[/Script/Survios.SVRLocalPlayerSettings]
SyncedPlayerSettings=(ControllerMode=Motion,HMDMode=HMD,HMDType=Survios,HandTrackerParent=Head,HandTrackerBaseOrientation=(RelativeLocation=(X=66.143242,Y=-5.098728,Z=48.728882),RelativeRotator=(Pitch=77.475006,Yaw=-8.435512,Roll=-5.044755)),AvatarCollisionMode=IgnoreBodyAndInteractionCollisions)
MotionControllerSettings=(MotionControllerThumbPadMode=Analogue,MovementDeadZone=0.100000,ThumbPadMaxVal=0.800000,InitialMovementDelayZone=0.800000,InitialMovementDelayMaxDuration=0.200000,InitialMovementDelayMaxThumbMovementDistance=0.300000,bNormalizeThumbPadInput=False,bSetInitialTouchLocationAsJoystickOrigin=False,bInitialMovementDelayIsEnabled=True,bRequiresThumbstickClickForMovement=True)
FirstPersonAvatarMode=ArmsOnly

betavr, 2016-07-17:
Can you provide more detail on why it didn't help? Did the game detect the hydra without the sixense dlls?

I'm trying to figure out if we can disable their Hydra driver completely, otherwise you are relying on their implementation, and I'm not sure I can help you with that. Although that SVRLocalPlayerSettings does look like it contains some kind of calibration data.

SooSiaal, 2016-07-17:
yes it detects the controllers without the dlls but they still function the same as before,before pushing any button or trigger the wands ingame follow the hydras,but as soon as a button is pressed the ingame wand moves off to a random position (usually infront or above you) and is turned 180 degrees

betavr, 2016-07-17:
Strange. Can you check if the Thirdparty\OpenVR directory contains a driver subdirectory?

SooSiaal, 2016-07-17:
only the OpenVRv0_9_12 folder with the api dll

frictionjones, 2016-07-21:
Hello, I just realized that the right mouse button does the same thing the right hydra trigger *should* do in raw data; is it possible to map the right mouse button to that trigger?

AlexLogin0ff, 2016-07-21:
Here in the video is the answer to your question - https://www.youtube.com/watch?v=p8ghvNIzWiw

betavr, 2016-07-21:
Nice video! It looks like hemisphere tracking is not being enabled by the game's builting driver.

betavr, 2016-07-21:
@SooSiaal Have you tried to disable/uninstall the hydra driver before starting raw data?

Btw the steam app will be online very soon, so the install/uninstall procedure will be much faster.

SooSiaal, 2016-07-21:
ohh no i havent tried that yet hehe

frictionjones, 2016-07-21:
Which hydra driver should be disabled or uninstalled?

AlexLogin0ff, 2016-07-21:
drivers do not need to be removed

AlexLogin0ff, 2016-07-21:
[image]

betavr, 2016-07-21:
@frictionjones This driver, but @AlexLogin0ff just clarified that Raw Data works for him without removing it.

betavr, 2016-07-21:
@AlexLogin0ff so the base station should be placed where the playarea center is?

AlexLogin0ff, 2016-07-21:
Yes!!!

betavr, 2016-07-21:
Awesome, can you guys please verify this? 

Also, is anyone here interested in betatesting the driver on Steam?

AlexLogin0ff, 2016-07-21:
Of course!

SooSiaal, 2016-07-21:
Sure thing,and it is somewhat working now,atleast much better than it was,gonna need a little finetuning with centering I guess,but this is promising atleast :)

betavr, 2016-07-21:
Thanks guys, I really appreciate your help! Please write to hydradriver@betavr.com to receive your beta key.

frictionjones, 2016-07-22:
Can you give step by step on how you got it working?

rblostwolf, 2016-07-22:
once calibrated does the hydra base station need to remain at shoulder height?

Honestydeceit, 2016-07-22:
Sorry, this just isn't working. Putting the base in the center of the room does nothing and putting the controllers next to the base does nothing, they just sit there and when picking the controllers back up, the hands are still fucked up. Honestly, it's completely unclear in the video just what this guy did to get it working, and I have no idea how to even start.

rblostwolf, 2016-07-22:
he did the room calibration and when the steam thing asks for you to hold your head set in the centre of the room he also put the razer base station at shoulder height in the centre of the space. i'm not sure whether leaving it there at shoulder height is required after calibration  if it is i'm not going to do it as it would get in the way

AlexLogin0ff, 2016-07-22:
Hydra station is put in the center of the square after installation Room SetUP!

frictionjones, 2016-07-22:
Yes, I calibrated with the base station in the center of the space (at shoulder height), and it still doesn't work. Pressing either of the triggers sends the hands into the air, and putting the controllers on the base station and picking them up again changes nothing.

AlexLogin0ff, 2016-07-22:
Further necessary to adjust the calibration in the game!

frictionjones, 2016-07-22:
When you say "adjust the calibration," you're talking about putting the controllers next to the base, right? I've done that many times and nothing changes once I've picked up the controllers again. The hands are still in the air and screwed up.

AlexLogin0ff, 2016-07-22:
"height" in the game!

[5 more replies]

closed 2018-08-13 by betavr