betavr/steamvr_driver_hydra #29: Update to OpenVR 1.0.6

betavr, 2017-03-30:
Started working on it.

BilboBaggins6704, 2017-04-03:
I'm just wondering, will there, will this add support for Oculus controls, like the X, Y, A, B controls? Because there would be so many games that this would be useful for.

betavr, 2017-04-05:
Yeah, maybe it can help with that.

cj7hawk, 2017-07-13:
Not just games. Just tried to load Google Earth VR, and while in Steam VR, I see Vive controllers, but Earth now detects the oculus for DK2 and defaults to Touch control.. And unfortunately, you can't tilt the image without the A/B buttons. Some kind of button mapping is going to be a necessity or else all the new games will probably just override our settings and we won't be able to play them - :( Fantastic driver BTW - Since I always VR from sitting, I really like the precision and response from the Hydra.

closed 2018-08-13 by betavr