betavr/steamvr_driver_hydra #31: Controllers work in game, not in the Steam System UI

sebkajeka, 2018-07-24:
I'm sure this used to work 4-5 months ago when I tested it. I can see and use the controllers in Steam Home but not in the Steam VR system menu. I don't have any laser pointers to interact with menus

betavr, 2018-08-13:
I noticed it too. Apparently SteamVR has been changed in many important ways in relation to the input system since the last release of this driver.

I'm trying to fix it, please be patient.

@JoeLudwig

betavr, 2018-08-16:
News break. 

The upgrade to OpenVR 1.0.16 is finished, the driver has its own controller bindings profile now in SteamVR. The profile still needs some work, but I can already use the hydra on the dashboard - laserpointers and buttons work as expected finally. Once that's finished I'll push out an update on Steam, probably this week if nothing goes wrong.

critchi, 2019-09-03:
Hi. Thank you very much for this.

About the new driver, released as beta on steam, I managed to configure the bidings for Steam Home, and it works perfectly.

critchi, 2019-09-03:
Hi. Thank you very much for this.

About the new driver, released as beta on steam, I managed to configure the bindings for Steam Home, and it works perfectly. the hydras appear and behave as expected.
For other apps though, different behaviors.

Google Earth:
You can't see the controllers at all, bindings don't work. If i get out of driver beta, then it works

critchi, 2019-09-03:
Hi. Thank you very much for this.

About the new driver, released as beta on steam, I managed to configure the bindings for Steam Home, and it works perfectly. the hydras appear and behave as expected.
For other apps though, different behaviors.

Google Earth:
driver beta: You can't see the controllers at all, bindings don't work. If i get out of driver beta, then it works as expected, with default bindings

The Lab:
Beta: At first the controllers didn't appear, and I wasn't managing to save the bindings. I finally managed to save the bindings, you have to attribute all available commands to some button to be able to save.
after that, the controllers appeared in app, the binding works, but the controllers appear upside down, making them hard to use. 
out of beta: controllers don't show up anymore. (they used to work in this app).

Other apps, like the spider man far from home vr, it doesnt work in any mode, beta, out of beta, with bindings, etc.

Please do share any news and thanks again for this. this is the only way I can experience vr today

r57zone, 2020-01-02:
In games, it also seems to have broken, for example, in PavlovVR it stopped working. Valve again broke something on SteamVR, a disgusting standard, constantly broaken.

r57zone, 2022-09-20:
[image]
Added recognition of Hydra controllers as Index controllers, this gives a little more convenient control and better compatibility (for example, with RE VR mods).
The updated driver can be found [here](https://github.com/r57zone/Razer-Hydra-SteamVR-driver).

nerblock, 2022-09-20:
Great job! Unfortunately my hydra is problematic so I can't try it. But I can put up a beta branch for this version on Steam if you need it.

betavr, 2022-09-20:
Great job! Unfortunately my hydra has problems so I can't try it. But I can put up a beta branch for this version on Steam if you need it.

r57zone, 2022-09-20:
Thanks, that would be great

closed 2024-06-27 by betavr